InfiniTimeOrg/InfiniTime #1385: Build Errors with M1 Mac - unrecognized command-line option '-arch'

CCranney, 2022-10-22:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I attempted to build the project (no edits) as a test run and it failed, likely because I have an M1 Mac

### What should happen instead?

A build file should be created

### Reproduction steps

As stated above, I'm reasonably sure the error is because I am using an M1 Mac. I was wondering if any developers have encountered this problem and how they worked around it. At the very least I thought you may want to update the `InfiniTime/doc/buildAndProgram.md` write-up with a warning if no solution is determined/exists.

I followed the instructions provided in `InfiniTime/doc/buildAndProgram.md`, specifically:

1. Downloaded dependencies, including the indicated cross-compiler, NRF52 SDK 15.3.0, python modules, cmake, and lv_font_conv. Note that this error likely stems from the cross-compiler.
2. Entered the following code:
```
git clone https://github.com/InfiniTimeOrg/InfiniTime.git
cd InfiniTime
git submodule update --init
cmake -DARM_NONE_EABI_TOOLCHAIN_PATH=[path to cross-compiler in Application folder]/12.2.mpacbti-bet1/arm-none-eabi -DNRF5_SDK_PATH=[path to unzipped download from the provided link]
make -j pinetime-app
```
This fails to build anything, and the following error occurs frequently in the output:
`arm-none-eabi-gcc: error: unrecognized command-line option '-arch'; did you mean '-march='?`

### More details?

I found the [following issue](https://gitlab.kitware.com/cmake/cmake/-/issues/23105) on a different repository that highlights the same error. However, my attempts to incorporate possible fixes haven't fixed the issue. This includes inserting the following text to the `CMakeLists.txt` file as they discussed in that forum:
```
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_VERSION "0.0.0")
```

I'm less than familiar with building projects this way, and I can tell it's showing here. I assume I need to update the `CMakeLists.txt` file or something similar, but am not sure where to make those changes.

Looking for an alternate way to resolve the error, I tried downloading `arm-none-eabi-gcc` through alternate means and using those instead. This includes downloading through xpm, addressing M1 Macs [here](https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/issues/12), which links to this download process [here](https://xpack.github.io/blog/2021/12/07/arm-none-eabi-gcc-v10-3-1-2-3-released/). I also looked at [this github repository](https://github.com/ARMmbed/homebrew-formulae) for downloading it with homebrew. In both cases I think the expected paths are different - rather than the above `unrecognized command-line option '-arch'` error, I get simple "cannot find file" errors. It's possible I just set the path incorrectly, but wasn't sure how to go about finding the exact path necessary to get the build to work.

### Version

v1.11 (latest)

minacode, 2022-10-22:
The person in #1365 and #1368 was able to build using podman and the docker image on an M1 Pro. I know this does not fix your issue but it may help you.

FintasticMan, 2022-10-22:
I've noticed that in your cmake command you are using a newer toolchain than the docs specify. Those haven't been tested thoroughly, and might not work. Could you try using version 10.3-2021.10?

Another thing, the `ARM_NONE_EABI_TOOLCHAIN_PATH` variable expects just the path that the toolchain has been decompressed to, not a subfolder of it.

I'm not really very well-versed in macOS though, and this is just general troubleshooting info, so I hope it is able to help anyway.

CCranney, 2022-10-22:
I tried using version 10.3-2021.10, but after getting it to run properly ran into the same error. It looks like `podman` may be the path forward, though I'm running into new problems getting that set up (see [this issue](https://github.com/containers/podman/issues/16262) on their GitHub). I don't know if @arduino-man ran into similar errors, I may reach out to them depending on how that issue unfolds.

While unresolved, there's likely nothing to be done on the InfiniTime repository (aside from possibly updating the build documentation). Should I close this, but update it as the podman issue progresses? Or would you prefer to keep it open while updates are expected?

minacode, 2022-10-22:
Just want to add that you may not need to specifically use podman. Docker should also do it. There is even [documentation](https://docs.infinitime.io/en/latest/build-flash-and-debug/index.html#using-docker) for it 😉

CCranney, 2022-10-22:
Docker, too, is being troublesome. This is just... one of those days where nothing works the way you want it to 🙄 it's funny because the apps I want to design aren't terribly complicated, but that's life I suppose

CCranney, 2022-10-23:
I finally got docker working! It was a privacy setting I had that was killing the connection. I'll keep the `podman` issue alive, but I'll close this issue. I suspect anyone with an M1 Mac will likely need to take the docker/podman route. Thank you for your help!

closed 2022-10-23 by CCranney

GottemHams, 2023-10-30:
There seems to be a problem specific to macOS where **something** causes `-arch` to get appended to the compiler arguments. I didn't really want to set up Docker just to compile something, so I decided to look into the issue.

Apparently simply adding `-DCMAKE_SYSTEM_NAME=Darwin` to your `cmake` command makes it work. ._. I think that without this, it decides the compiler should target the host's own architecture. In my case the error was about `arm64` (which corresponds to Apple's M1 chip), while the toolchain should target **32-bit** ARM. By explicitly setting `CMAKE_SYSTEM_NAME` this autodetection no longer happens and the build process continues by using what is **actually** set in Infinitime's own configs.