InfiniTimeOrg/InfiniTime #1388: Analog Watch Face Slow Animation

antonlinux123, 2022-10-23:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

When you swipe back to the home screen from the left side menu a very slow animation plays.

### What should happen instead?

The animation should be much faster.

### Reproduction steps

https://user-images.githubusercontent.com/116518833/197414629-5b53dc47-8fb2-4b81-8dc0-eaa1d7d0f553.mp4

### More details?

I don't know anything else about the bug.

### Version

v1.11.0

Avamander, 2022-10-23:
Can you please determine that it was faster before?

antonlinux123, 2022-10-23:
I am not saying that it was faster before I am trying to say that the animation is slower then it should be to feel comfortable this might just be preference though.

RageGamerBoi, 2022-10-24:
This is because the Analog watch face is very asset heavy and all watch faces need to load every time they’re swiped to.
This results in it loading slower then the ones made with simple rectangles and fonts.
It feels wrong but it’s totally expected.

Avamander, 2022-10-25:
There are very few ways around that. With effort someone might be able to optimise this but right now it's expected and not really a bug.

closed 2022-10-25 by Avamander