InfiniTimeOrg/InfiniTime #1389: Default PineTime Style Colors

RageGamerBoi, 2022-10-24:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Cante the default PTS color scheme to slightly more pleasing colors.

### Description

The current default monotone accent on the PTS face is a little bit bland, maybe change the default to something with a little more contrast and color like this:
[image]
It’s a small and (hopefully) easy change to make but it make the watch face a little bit nicer.

Riksu9000, 2023-01-02:
I think I'd prefer this one. What do you think @kieranc?

kieranc, 2023-01-02:
[image]

Like this? I'll try it out on device today. I don't have a strong opinion about it tbh, I've spent a lot of time staring at the default colors, usually in the sim, and on my watch I have it set it whatever colors I feel like at the time... I think if the colors were to be overhauled, it would make sense to get input from as many users as possible to find which color schemes are popular and then implement them as a list to replace the Random button, which isn't very random at all.

NitroNils, 2023-01-05:
Quick input:

I use this face at night with red colors for protecting my "night vision" / for sleeping better.

RageGamerBoi, 2023-01-06:
This would only be changing the default colors to something white on black for better default readability with an accent color that is tbd.

@kieranc I used the light teal (I think it looks the nicest and is the most usable/readable), unless the simulator colors are just buggy.

fossison, 2023-03-25:
If the intention is for defaults: maybe orange to match the InfiniTime logo?

Orange sidebar and digits:
[image]

Orange sidebar and white digits:
[image]