InfiniTimeOrg/InfiniTime #1390: Return to Clock A Few Seconds After Display Sleep

RageGamerBoi, 2022-10-24:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

About 10-15 seconds the display goes to sleep, if the watch has a navigation screen up (ex. Quick settings) it should return to the time.

### Description

If a navigation screen is opened and the watch goes to sleep because it isn’t being used, it should return to the main time display. This shouldn’t apply inside apps or for a few seconds after the watch goes to sleep because the page might need to be used again.
This is useful when a menu screen is left open (like the notifications shade etc.) because you forget to close it and you go to quickly check the time later and you’re greeted with the screen you had open and have to fumble with the watch to check the time.

minacode, 2022-10-25:
I disagree (a little). Personally, I send the watch to sleep by pressing the button a few times until it sleeps. Whenever I let it sleep on a screen I intend to use it later again. It's mostly the timer that I want to check quickly. 

On a less personal level, that behavior would erase any unsaved in-app state. Think of Twos for example. 
Given the PR that implements the app stack, we should maybe aim for persistence implementations for every app 🤔

Riksu9000, 2023-02-21:
A similar idea was previously closed as wontfix #570, however considering the navigation screens only it might not be a bad idea. With this change, quick settings and the app list would sort of be a part of the same "app" as the watch face conceptually.

Is this a common enough problem though that it is worth implementing this feature, even if #618 and #1592 were fixed?

RageGamerBoi, 2023-02-21:
This issue was originally trying to fix my own problems relating to #618 , regardless of touch driver problems etc.
I believe this should probably still be implemented in some fashion as even accidental touches cause problems, and seeing as the PineTime is primarily a watch, it seems like a logical solution.
Having the time in every app is a different discussion imho as a small clock in the corner is easy to miss at a glance, but idk.

RageGamerBoi, 2023-02-21:
I have put this suggestion as a reply on #618

closed 2023-02-21 by RageGamerBoi