InfiniTimeOrg/InfiniTime #1391: Add Scroll Bar to the Music App

RageGamerBoi, 2022-10-24:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

The music app has almost hidden volume buttons, and it seems like a good idea to add some sort of indicator to show that they exist.

### Description

A recent pull request reminded me that this is a problem:
https://github.com/InfiniTimeOrg/InfiniTime/pull/1386#issuecomment-1289000691  & https://github.com/InfiniTimeOrg/InfiniTime/pull/1386#issuecomment-1289407562

Spagett1, 2023-01-30:
yeah i only found out the volume functions by accident, some sort of indicator would be good

Findirfin, 2023-06-10:
I found out about them by reading this, I was about to make my own custom volume app of some sort, so I'm glad I stumbled upon this. There for sure needs to be some sort of indicator.

LinuxinaBit, 2025-11-04:
Glad that this was fixed!
Recently I’ve come across https://github.com/InfiniTimeOrg/InfiniTime/pull/2337#issuecomment-3221847165 and I think it is a better solution to the actual root problem of volume shortcut accessibility.
Hope to see it merged eventually.
Until then, I’m glad we have a serviceable interim solution.

mark9064, 2025-11-04:
Yeah definitely, #2337 hasn't been abandoned, this PR was merged as an improvement over what we had before and is not the end of work :)