InfiniTimeOrg/InfiniTime #1393: Morse - started, but can't add another app.

programmin1, 2022-10-25:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Morse added as app

### Description

I thought it would be great to have some morse app, to tap morse in to decode it and to list out all the morse dot dashes for letters at the very least. However when I get everything but the app main screen icon it compiles, when I add the main screen icon

    {".", Apps::Morse},

It gives an error like the array reached some sort of limit now...
/InfiniTime/src/displayapp/screens/ApplicationList.cpp:1:
/home/luke/Projects/InfiniTime/src/displayapp/screens/ApplicationList.h:56:10: error: too many initializers for ‘std::__array_traits<Pinetime::Applications::Screens::Tile::Applications, 12>::_Type’ {aka ‘Pinetime::Applications::Screens::Tile::Applications [12]’}
   56 |         }};
      |          ^

https://github.com/InfiniTimeOrg/InfiniTime/pull/1392

Avamander, 2022-10-25:
Have you tried changing the size of that array to accomodate for an another element?

FintasticMan, 2022-10-25:
See the comment I left on your pull request.