InfiniTimeOrg/InfiniTime #1394: InfiniTime 1.11.0, Gadgetbridge 0.71.3/0.71.2/071.0, Resources file "Element cannot be installed"

phalkon13, 2022-10-26:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

InfiniTime 1.11.0, GadgetBridge 0.71.3/0.71.2/071.0, Resources file "Element cannot be installed"

### What should happen instead?

file installing

### Reproduction steps

Tried 3 versions of Gadgetbridge, all perform the same way, as well as tried downloading the file multiple times.

### More details?

I just recently updated my PineTime from 1.0.0 to 1.11.0, which seems to be working just fine, and the infinitime 1.11.0 update pushed just fine, and I can re-push it just fine, but the resources file fails immediately.
File is named "infinitime-resources-1.11.0.zip", and shows to be 15.92k on my phone.
I have a "Samsung Galaxy S21 Ultra 5G" running Android 12 (everything is as current on the stock OS as it can be).

### Version

1.11.0

### Companion app

Gadgetbridge 0.71.3/0.71.2/071.0

Elements6007, 2022-10-26:
I dont think GadgetBridge can upload the resource files. According to the documentation only Amazfish and ITD have the functionality needed. [documentation](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/gettingStarted/updating-software.md#updating-resources)

JF002, 2022-10-26:
I mentioned it in the doc and in the release note : as of now, only [ITD](https://gitea.arsenm.dev/Arsen6331/itd) and [Amazfish](https://github.com/piggz/harbour-amazfish) have integrated the support for the new resource package. Let's give the developers of the other companion apps to integrate it as well ;-)

Regarding Gadgetbridge, I have already [opened a feature request](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2919). Wait and see :)

I'm closing this issue as this is not an InfiniTime bug.

closed 2022-10-26 by JF002

phalkon13, 2022-10-26:
Ok thanks

FSMaxB, 2023-09-04:
Do I understand that correctly: There is no way to update the resource file from anything other than from desktop Linux and SailfishOS at this point in time?

Also the resource files are kind of essential for InfiniTime updates, are they not?

FintasticMan, 2023-09-04:
The resource files can be uploaded from android, but it has to be done manually using a tool like [InfiniTimeExplorer](https://infinitimeexplorer.netlify.app/). You need to copy the files in the zip to the directories specified by the json file. The resources aren't strictly necessary for InfiniTime to work, it just means you miss out on some features that need them. These could be things like watch faces or apps.