InfiniTimeOrg/InfiniTime #1399: New 1.11 watchfaces greyed out

wantyapps, 2022-10-29:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I downloaded the DFU zip, installed it with the InfiniLink app, tried to set a new watch face, and the new two were greyed out and un-clickable.

### What should happen instead?

I should have been able to set a new watch face from the new release (InfiniNeat, G7710)

### Reproduction steps

1. Install the new firmware from a local file
2. Try to set a new watch face from the new release

### Version

v1.11

### Companion app

InfiniLink from the App Store

Avamander, 2022-10-29:
You need to install the resource pack required by those watchfaces.

closed 2022-10-29 by Avamander

wantyapps, 2022-10-29:
Oh! Thanks! :) have a good day

wantyapps, 2022-10-29:
One last question, if you don't mind: how do I flash the resources with InfiniLink? I tried the normal firmware flash option, but,as expected, "The file is not a DFU zip". Did I miss something in the app?

Avamander, 2022-10-29:
@wantyapps InfiniLink currently does not support resource pack upload.

wantyapps, 2022-10-29:
Got it, thanks for the help! Will upload on my computer when I get close to it.