InfiniTimeOrg/InfiniTime #1400: LUA Scripting support

agittins, 2022-10-29:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Could InfiniTime include a lua scripting engine?

### Description

Is it feasible to include a lua scripting engine into infinitime? I don't know anything about the technical aspects of implementing it, only that I have seen a few projects on what I assume is limited hardware (eg: MagicLantern, running on Canon cameras) providing surprisingly capable lua api's. With sufficient work in the api it's possible to do fairly low-level stuff with reasonable efficiency, and for scripts to be loaded/unloaded at runtime on demand.

It strikes me that this would allow for:
- lower barrier to entry for people who want to do things like custom watchfaces or other apps without having to learn c++.
- a way to rapidly explore prototypes or test hardware interfaces (eg: exploring self-calibration algorithms for the heartrate sensor would be easier by iterating on a script (for someone without solid c++ knowledge).
- using lua may provide an "easier"(?) way to load/unload software features without the complexities of Position Independent Code (PIC) when trying to load compiled c++ "modules".

As I said, I don't know the technical difficulties that might face such a goal, so there might be dead-simple reasons why this isn't practical - in which case I'd welcome them being spelled out here so that this issue can be closed and serve as an FAQ for the query. Alternately, maybe it's a job only restricted by someone with time+knowledge, in which case this issue can serve as an advertisement to catch anyone skimming around for their next passion project :-)

Please note I am not asking for any one person's opinion if they think lua would not be useful to them personally, more so if it is technically feasible to implement, and what the realities of doing so might involve. Since there are no issues already mentioning lua I assume it's either an obvious answer (like it requires an mmu feature or some other hardware that's lacking) or nobody's thought to ask outside of chat.

Avamander, 2022-10-29:
Any interpreted language is quite heavy, probably too heavy in this context. Learning C++ to a sufficient level for creating watchfaces is not that difficult. Using the hardware SWD interface or the simulator already provides fast prototyping.

An interpreter is unlikely to be added (any time soon).

closed 2022-10-29 by Avamander

agittins, 2022-10-29:
"Any interpreted language" seems a *very* broad qualification, but I appreciate the quick and unambiguous response. Cheers!