InfiniTimeOrg/InfiniTime #1404: Unable to flash infinitime with openocd

RazRosman, 2022-10-30:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Error Message When running openocd

### What should happen instead?

I should see a connection to the pinetime

### Reproduction steps

I installed openocd following the manual then I ran:
openocd -s /home/pi/openocd-code/tcl -f ../tcl/interface/rpi3.cfg -f ../tcl/target/nrf52.cfg
and I've got the following error:

Debug Adapter does not support any transport? check config file order
unable to select a session transport. can't continue.

rpi3.cfg looks like this:
adapter gpio swclk 23
adapter gpio swdio 19
adapter gpio srst 18

### More details?

I'm using rpi3 and connecting to pinetime with the pins directly from the rpi3 with no st-link

### Version

I cloned it from this repo

Avamander, 2022-10-30:
You have to have SPI enabled on your Pi, but that aside, it's not really an InfiniTime issue.

RazRosman, 2022-10-30:
So, SPI is enabled and I got the same error.
I'm trying to configure openocd again with --enable-sysfsgpio --enable-bcm2835spi
The thing is that I'm following the exact steps in the guidelines but I don't have a STLINK so I'm kindda stuck

RazRosman, 2022-10-30:
I managed to solve it.
I installed pinetime-updater from this link
https://github.com/lupyuen/pinetime-updater

and then used the bin file I built based on this article and it worked.

I can see my app on Pinetime.

Thank you very much

closed 2022-10-30 by Avamander