InfiniTimeOrg/InfiniTime #1405: Increase the heart rate data logging interval

stevensu1838, 2022-10-31:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Increase the heart rate data logging interval

### Description

I've noticed that the heart rate on Pine Time watch is updated every few seconds. How can I increase the heart rate data logging interval to collect more heart rate data in the same time period? Like how can I make the heart rate data update every second? Thanks a million. Cheers.

JF002, 2022-11-05:
Are you asking this question as a user or as a developer?

The update interval is caused by how the current algorithm is designed. As a user, you cannot increase or decrease it, InfiniTime does not provide this functionality (yet?). As a developer, you'll need to improve the HR algorithm so that it yields measurements more often. Your contribution would be more than welcome ;-)

stevensu1838, 2022-11-11:
@JF002 Hi, Thank you so much. Do you know what exactly the heart rate data logging interval is? Can you please show the heart rate interval part of the current algorithm? I am a newbie but would love to explore and contribute. Cheers

JF002, 2022-11-14:
@stevensu1838 According to the code, we run the heart rate task every 40ms when the data acquisition is enabled (https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/heartratetask/HeartRateTask.cpp#L32).

Then, the PPG algo returns an actual heart rate value when is has accumulated 200 samples (https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/components/heartrate/Ppg.cpp#L61), which means that a new value is available every ~8s.

Reducing those constants will probably decrease the accuracy of the measurements, though.

from-nibly, 2023-03-21:
Just a thought, the way you'd probably need to accomplish more frequent measurements is not through shorter samples. but rather a sliding window of samples. if you take 200 samples, you can update the "measurement" by adding a 201st sample and only including the most recent 200 samples in your math. That way you'd be able to have a heart rate measurement every 40ms (after the first 8 seconds that is)

JF002, 2023-03-25:
@from-nibly #1486 will improve the accuracy and speed of measurement :)

closed 2023-04-30 by JF002