InfiniTimeOrg/InfiniTime #1409: Preset Timers

tweak42, 2022-10-31:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Timer start screen with multiple preset timers

### Description

Switch the default timer screen with 1,5,10,30 and Custom minutes buttons.
The number buttons in a 2x2 grid, and the "Custom" full width wide at the bottom of screen.
Each of the numeric buttons would start the timer with it's allotment of time.
The custom button would launch the present timer screen that is increment able.

RageGamerBoi, 2022-11-02:
It’s a little complicated and would clutter up the timer interface significantly.
Good idea though 👍 something similar could be better implemented  in the future.

zgypa, 2022-11-05:
I would also use this feature a lot.

Riksu9000, 2023-02-21:
The issue here being the time setting being cumbersome, this is essentially a duplicate of #671.

closed 2023-02-21 by Riksu9000

margamanterola, 2024-07-07:
> The issue here being the time setting being cumbersome, 

Not really. Even if setting the timer is less cumbersome, it's always going to require some work. Having a few presets would go a long way to making usage of the timer app a lot better. Particularly for folks that typically use the same values.

My idea would be:

1. First screen of the timer app shows 6 bubbles with 5 presets and a "+" icon for custom timer.
  - The presets can start with some defaults like 5, 10, 15, 20, 30 minutes.
  - But then they get updated with the user's usage.
  - The settings for the app store the last 5 times used.
2. Pressing the + icon takes you to the current main screen, i.e. creating a new timer.

I'm willing to create a PR to show how this would work. Could we re-open this issue?