InfiniTimeOrg/InfiniTime #1412: measure and display heart rate variability (HRV)

khimaros, 2022-11-01:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

heart rate variability (HRV) measures the dynamic range of your heart

### Description

according to https://my.clevelandclinic.org/health/symptoms/21773-heart-rate-variability-hrv

"If your heart rate is highly variable, this is usually evidence that your body can adapt to many kinds of changes. People with high heart rate variability are usually less stressed and happier."

"In general, low heart rate variability is considered a sign of current or future health problems because it shows your body is less resilient and struggles to handle changing situations."

minacode, 2022-11-02:
Sounds interesting but the article you provided says
> The majority of wrist-worn fitness devices and trackers track your heart rate through your skin. Unfortunately, this means they usually aren't sensitive enough to detect heart rate variability accurately.

khimaros, 2022-11-10:
it looks like @Pittconnect made some degree of progress doing HRV analysis with wasp-os and HeartPy: https://github.com/daniel-thompson/wasp-os/issues/273#issuecomment-1002804919 -- here's a video of their attempt: https://www.youtube.com/watch?v=FsX2-1JJF6o

there is some additional detail about HRV and its health implications here: https://support.ouraring.com/hc/en-us/articles/360025441974-An-Introduction-to-Heart-Rate-Variability

i think my first step will be getting a few minutes of raw PPG data from the watch and do a bit of offline analysis to see if the signal is clean enough to be useful.

minacode, 2022-11-11:
This sounds way more promising!

> i think my first step will be getting a few minutes of raw PPG data from the watch and do a bit of offline analysis to see if the signal is clean enough to be useful.

Good luck! Would be nice if you succeed.