InfiniTimeOrg/InfiniTime #1414: Notification spam on new music

HugoMskn, 2022-11-01:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Everytime a new music is played with spotify, I get spammed of empty notification

### What should happen instead?

Either a notification telling the song name or nothing

### Reproduction steps

Connect your watch to your phone (android for me) 
install gadget bridge 
Install spotify and wait until the song switch, then you should get spammed

### More details?

I dont know if it's a gadget bridge related issue, but maybe make so that empty notification arent displayed ?

### Version

1.11.0 

### Companion app

Gadget bridge V0.71.3

Avamander, 2022-11-01:
Add Spotify (back) to Gadgetbridge's notification filter list.

JF002, 2022-11-05:
I'm using Gadgbetbridge 0.69.0 and never notice that issue... I'll probably have to update to the new version of Gadgetbridge to check. 
In the mean time, I would be interested to know if you still have the issue if you downgrade your Gadgetbridge to 0.69. That would help to see if the issue is caused by InfiniTime or Gadgetbridge.

Boteium, 2022-11-25:
no issue on my system.

GadgetBridge 0.71.3
InfiniTime 1.11

dozadoesit, 2023-03-24:
I do get spam occasionally from other things and as others have said it's pretty much a gadgetbridge bug if anything. I think if you uninstall the app reinstall it, add pinetime again and it keeps happening check with the gadgetbridge community and file a bug with them.

JF002, 2023-03-26:
> I'll reopen this issue until we check that this issue is not caused by a potential InfiniTime bug :)

I use Spotify everyday with my PineTime connected to the phone (running gadgetbridge 0.73.0 right now) and I've never noticed this issue either.

I don't know what's causing this issue, but I really doubt this is a bug in InfiniTime.

closed 2023-03-26 by JF002