InfiniTimeOrg/InfiniTime #1419: Steps not counting

Gernomaly, 2022-11-06:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

It seems like after I updated to 1.11.0 my watch stopped counting the steps. Even after rolling back it is not working anymore

### What should happen instead?

It should have continued to count the steps

### Reproduction steps

I only have this one watch so I can not try to reproduce it on another watch. 

### More details?

I rolled back, tried it on the old version and tried to flash the update again. Even with different Companion Apps. But it is not working anymore.

Could it be a hardware issue?

### Version

1.11.0

### Companion app

InfiniLink and Gadgetbridge for the Updates. ITD for the ressources.

pptime02, 2022-11-06:
So did it not work after you rolled back to 1.10 as well? You could also try downgrading to 1.10 by re-flashing it.

To rule out a hardware issue: does the "Motion"-App (Charts-Icon) still work? And does "Shake wake" still work?

Gernomaly, 2022-11-06:
> So did it not work after you rolled back to 1.10 as well? You could also try downgrading to 1.10 by re-flashing it.
> 
> 
> 
> To rule out a hardware issue: does the "Motion"-App (Charts-Icon) still work? And does "Shake wake" still work?

Both of them are not working anymore. I am trying to reflash to 1.10 when I am at my pc later. 

Is there a way to remove the ressources as well?

NeroBurner, 2022-11-06:
Seems like a hardware failure. Last thing you can do is to completely drain the battery and then recharge to truly power cycle the hardware. Easiest way is to enable heart rate monitoring and activate the flashlight app

Gernomaly, 2022-11-09:
> Seems like a hardware failure. Last thing you can do is to completely drain the battery and then recharge to truly power cycle the hardware. Easiest way is to enable heart rate monitoring and activate the flashlight app

It sadly looks like it is a hardware failure and power cycling the hardware did not manage to fix that. 😢

closed 2022-11-09 by Gernomaly