InfiniTimeOrg/InfiniTime #1424: Notifications doesn't support Scandinavian characters æ,ø and å

dvidebaek, 2022-11-07:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Support characters æ, ø and å

### Description

Notifications messages doesn't support the Danish/Skandinavien characters æ, ø and å. A word like påvirket (affected) would be spelled pvirket in the notification message. The lettet å is omitted instead of displayed.

minacode, 2022-11-07:
Related to #212.

yusufmte, 2022-11-08:
If you happen to use GadgetBridge as your companion app, there is a transliteration setting for many languages including the three letters above :) So it will transliterate the notification before passing it to the PineTime.

The transliteration file is [here](https://github.com/haditim/Gadgetbridge/blob/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/util/LanguageUtils.java).

The letters you ask about are transliterated as follows:
```
// Scandinavian characters
put('Æ',"Ae"); put('æ',"ae");
put('Ø',"Oe"); put('ø',"oe");
put('Å',"Aa"); put('å',"aa");
```

You can enable it by going to GadgetBridge, click on the settings wheel that is under your connected PineTime (not the GadgetBridge settings), then click on transliteration settings.

I use this feature for arabic and find it a good solution.

closed 2022-11-18 by Avamander