InfiniTimeOrg/InfiniTime #1425: Display turns on every 30 min when Shake Wake is active

efelon, 2022-11-07:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

While Shake Wake is active, the display turns on automatically every full and half hour.

### What should happen instead?

I expect that the display stays off unless there was a shake.

### Reproduction steps

* Turn on Shake Wake
* Observe the display every 30 minutes to the full and half hour
  (you may place the watch on the table to prevent accidental shake activation)

### More details?

* Chimes are off.
* Other Wake Up options seam not to influence this behaviour.

### Version

v1.10.0 and v1.11.0

### Companion app

Gadgetbridge

JF002, 2022-11-19:
Could you try to enable and then disable the Chimes (enter the Chimes settings, enable it, exit the settings to save it and then disable it) and check if it fixes the issue?

efelon, 2022-11-19:
Yes indeed, that solves it but only until the next reboot/battery empty.

efelon, 2022-11-20:
Sorry, no it does not solve it. My previous test was while the watch was charging (where it does not turn on every 30 minuets). As soon as I put it back on my wrist the problem resumed. And I repeated  your requested procedure several times now, without success.

AtomToast, 2023-01-09:
I have the same issue. It always turns on at full and 30 past

segaretroboy, 2023-08-07:
I have been having this issue as well, only enabled wake action besides the button is to shake wake and the screen turns on every 30 minutes regardless of the chime setting.

mark9064, 2024-11-18:
This is probably influenced by motion being updated on every systemtask loop, meaning that when the onnewhour/halfhour events arrive, motion is calculated twice in quick succession causing the acceleration value to be calculated way higher than it should be. This issue will have been partially alleviated by #2041 but I will be opening constant frequency motion updates as a PR soon :)