InfiniTimeOrg/InfiniTime #1427: Ability to change step goal by 500 step increments

calhix, 2022-11-10:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

I'd like to be able to change the step goal by 500 increments. 

### Description

Ideally I'd like to set the step goal to 7500. I was doing some research and saw one study stating after 7500 steps a day, the benefits tend to quickly trail down. We could also think about changing the default to 7500 but 10k is the default nearly everywhere so it's not a big deal. 

Article: https://www.prevention.com/fitness/fitness-tips/a31788212/how-many-steps-should-you-walk-a-day/

minacode, 2022-11-10:
Your wish for 500-steps is valid. 
However I doubt you are an elderly woman that wants to do no more step than necessary to increase your 4-year mortality, so don't overvalue such numbers. I think the default 10000 are still a good default.

Dheenhasty, 2022-11-21:
Another way to improve this is to duplicate +/- and add +/- 1000 and +/- 100

minacode, 2022-11-22:
Honest question: do you trust the sensor enough that 100-granularity is needed for a whole day?

Dheenhasty, 2022-11-22:
Of Course !!!!! :) More seriously yes i think the error margin of 100 a day could fit. 

Just want to propose another solution to set up the step more easily, but the more i think of it the more i think the 500 step is ok on the watch (maybe adding a button to reset to default or static value ? ) 

the ideal solution could be a custom variable pushed directly thru companion app.

Riksu9000, 2022-12-30:
I don't know about increasing the granularity just to include a specific goal and making adjustment require twice the amount of presses, or clutter the UI. I doubt that 7500 is the precisely calculated perfect value, such that we must support it in particular.

calhix, 2022-12-30:
I don't care so much about changing the default value of 10k but I do think making the ability to change by 500 is good. The sensor IMO is good enough for it and I don't think the average person is going to be messing with their step goal often. I do agree the UI should remain the same to reduce clutter. Especially if merging the step goal quick setting and steps app.

Riksu9000, 2022-12-30:
Can you explain why someone would want to pick 6500 steps, and 6000 would not be acceptable?

calhix, 2022-12-30:
In the future if we have a goal reached notification as I brought up in #641 I'd like to know when I reached my preferred goal of 7500. Its not contingent on that issue though and there's no biggie if I go over of course but that's what I'd like my minimum steps a day to be and perhaps there are others as well.

PR #1575 (2023-02-12): Update setsteps UI
Closes #1427

Long pressing will change the value by 1000, whereas clicks will change
it by 500. This allows setting more precise values, while also making it
easier to set any value.

[image]
Ignore the fact that the animation is way too fast.
closed 2023-02-12 by Riksu9000