InfiniTimeOrg/InfiniTime #1432: 1.69 inch screen capacitive touch screen recommendation

modi12jin, 2022-11-13:
[image]

Official Website: www.polcd.com
Alibaba International Station Store: https://polcd.en.alibaba.com/

Recommended reason:
 Screen size 1.69 inches
 Resolution 280*240
 Screen: ST7789V (SPI)
 Capacitive touch CST816T (I2C)
 Touch and screen share one FPC
 Good quality and cheap, if you are in China it's worth 24 rmb

modi12jin, 2022-11-13:
Use this screen video

https://m.youtube.com/watch?v=2TOVohmUqOE&t=5s

modi12jin, 2022-11-13:
About touch drive, maybe this question can help you guys
https://github.com/fbiego/CST816S/issues/1

Avamander, 2022-11-13:
Hardware design of the PineTime is not really up to the InfiniTime project to decide upon. Feature and enhancement requests have currently been written down on the Pine64 Wiki article called "PineTime Hardware Wishlist."

closed 2022-11-13 by Avamander