InfiniTimeOrg/InfiniTime #1433: AddHUDsupport

modi12jin, 2022-11-13:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

I believe adding HUD support to the watch would be a really useful feature.  However, it requires the cooperation of the mobile terminal

### Description

This video demonstrates

https://www.youtube.com/watch?v=Kb0fgH8GiYA

Avamander, 2022-11-13:
We do have a navigation watchapp for that purpose already. Supported by itd and Amazfish with Puremaps.

closed 2022-11-14 by modi12jin

modi12jin, 2022-11-14:
@Avamander I want to run on Android and I don't have a SailfishOs phone.

JF002, 2022-11-14:
Amazfish and ITD run on any Linux distro. 

You can also have a look at this branch (https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/osmand-experiments) in Gadgetbridge that integrates osmand and the navigation API of InfiniTime.