InfiniTimeOrg/InfiniTime #1440: Customize vibrations on hour and half hour (avoid nights/weekends)

famiclone6502, 2022-11-14:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Only chime on weekdays and during the day

### Description

Chimes help break my focus during work hours and check for meetings I need to attend. Can you add a menu option to start and stop these during custom hours? Also, a check box for whether or not they occur only weekdays (Monday through Friday)? Otherwise it will keep me awake at night, and I do not need them during the weekend. Right now I need to disable the chimes manually on my PineTime.

I've been using a Timex Data-Link with a custom Motorola 6805 application to perform chimes on a specific schedule. Right now it does hourly chimes during the day (9 AM to - 6 PM) and only during weekdays. I do appreciate the half-hour chimes that the PineTime does already, which I wanted on the Timex.

Ideally, customization could eventually go even further. There's a lot of history in different types of chimes clocks/watches have offered like different repeaters and sonneries. Personally, I would like an additional reminder a minute before the hour or half-hour. Customizing with a shorter vibration for that, plus maybe an audible chime on the hour would be helpful. Others may also only want specific days of the week. These are not crucial to me but would be nice to consider in the design of related features. 

This request builds upon issue #778, which added the option for vibration on the hour and half hour. Without this feature I would not have bought the PineTime, so thank you for fulfilling this request.

minacode, 2022-11-14:
> Otherwise it will keep me awake at night

Sleep mode disable chimes. Maybe that it worth a try for you.

famiclone6502, 2022-11-15:
> Sleep mode disable chimes. Maybe that it worth a try for you.

From what I understand, it would still be a manual process to enter sleep mode. I also still want other notifications outside of work hours (i.e. during the weekend), just not chimes.

Boteium, 2022-11-25:
Automatic sleep mode sound awesome. It can also save battery during night.