InfiniTimeOrg/InfiniTime #1443: Could not find LV_FONT_CONV when building InfiniTime

Anonymous941, 2022-11-18:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

InfiniTime will not build

### What should happen instead?

InfiniTime building

### Reproduction steps

I'm trying to build InfiniTime.  I can build InfiniSim but when I try to make it to flash to a watch, it has an error.

```
PROJECT_GIT_COMMIT_HASH_SUCCESS? 0

BUILD CONFIGURATION
-------------------
    * Mode : Release
    * Version : 1.11.0
    * Toolchain : ../../gcc-arm-none-eabi-10.3-2021.10
    * GitRef(S) : 38092fcb
    * NRF52 SDK : ../../nRF5_SDK_15.3.0_59ac345
    * Target device : PINETIME
    * Debug pins : Disabled
    * Build DFU (using adafruit-nrfutil) : Disabled
    * Build resources : Disabled
CMake Error at src/displayapp/fonts/CMakeLists.txt:4 (find_program):
  Could not find LV_FONT_CONV using the following names: lv_font_conv
```

### More details?

I've tried everything: copying the src folder from InfiniSim, installing it in a bunch of different directories, running the command in different folders, but I always get this error.

### Version

v1.11

closed 2022-11-18 by Anonymous941

JF002, 2022-11-19:
@Anonymous941 I guess you've fixed this issue? It could be interesting to other contributors if you could explain how you fixed it :)

I guess you've successfully installed [lv_font_conv](https://github.com/lvgl/lv_font_conv) on your system ?
Also, remember that you can use the [docker container](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/buildWithDocker.md) to build the project if you don't want to install all the build dependencies on your system.

Anonymous941, 2022-11-19:
> @Anonymous941 I guess you've fixed this issue? It could be interesting to other contributors if you could explain how you fixed it :)
>
> I guess you've successfully installed [...]

It was answered on the PineTime matrix so I closed it here, it turns out my node modules wasn't in my path, so I added `~/.nvm/versions/node/v18.12.1/lib/node_modules` (the output of `npm root -g`) to my path and it worked.

Hope it helps anyone else who has this issue!

JF002, 2022-11-19:
@Anonymous941 Great, thanks!