InfiniTimeOrg/InfiniTime #1446: Sensors broken after upgrading to 1.11

Coleej, 2022-11-19:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

No movement or heart rate data is appearing.

### What should happen instead?

See x, y, and z movement and heart rate

### Reproduction steps

Upgraded to 1.11 using ITD compiled on Manjaro ARM. Not sure how to reproduce.

### Version

1.11

### Companion app

GadgetBridge 0.7.13

minacode, 2022-11-20:
What happens when you downgrade to a previous firmware like 1.10?

Coleej, 2022-11-21:
I downgraded to 1.10 and the sensors are still not working.

Coleej, 2022-11-21:
I downgraded to 1.10 and the sensors did not work. I also upgrade back to 1.11 without success.

Honestly, I don't know how long they've been unresponsive, I don't use their functionality that much.

spedicy853, 2022-11-21:
I had the same issue except the heart rate monitor still works. I upgraded to 1.11 with gadget bridge but installed the resources with itd.

Coleej, 2022-11-21:
I spoke too soon. The heartbeat monitor does indeed still work. I thought it would be on by default.

Cody Johnson
-------------------
t: 239 233 0155
e: ***@***.***
e: ***@***.***

-------- Original Message --------

JF002, 2022-11-25:
At first sight, it mostly look like a hardware issue rather than a software issue, unfortunately. I've seen mentions of dead motion sensor after a shock, or after the watch felt on the ground,.. Do you think it could be the case for your device?

spedicy853, 2022-11-25:
Possibly, it has been dropped, though I know it had been dropped and had stool worked at the time.

pptime02, 2022-11-27:
I think doing a full hardware reset by letting the battery run out would be worth a try.

spedicy853, 2022-11-28:
I will try and see what happens

spedicy853, 2022-11-29:
No luck, went to the sensor menu and there was no reading.

JF002, 2022-11-30:
I'm sorry to read that your issue is still not solved.
However, I don't think we'll be able to do anything at the software level to fix it so, if you don't mind, I'll close this post. 

I suggest you contact Pine64 directly to check if they can do anything about it.

closed 2022-11-30 by JF002