InfiniTimeOrg/InfiniTime #1448: Firmware update going at 200 bytes per second

Anonymous941, 2022-11-21:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The firmware update is going at an incredibly slow speed

### Reproduction steps

I have a Linux computer that I'm using to send a firmware update, and I've tried itd and Siglo but they both update the firmware at 200 bytes per second.  I've tried on an Android phone and it works much more quickly.  My computer has Bluetooth 4.0.  I'm testing custom firmware and need to flash it frequently so it's been annoying to have to wait around 15 minutes for the firmware to flash.

I think it has something to do with `src/drivers/Bma421_C/bma4.h` having multiple lines that include `BMA4_OUTPUT_DATA_RATE_200HZ` (lines 395, 1838 and 1883) but I don't know how to fix this.

### Version

1.11

Avamander, 2022-11-21:
> I think it has something to do with src/drivers/Bma421_C/bma4.h having multiple lines that include BMA4_OUTPUT_DATA_RATE_200HZ (lines 395, 1838 and 1883) but I don't know how to fix this.

Very unrelated.

> My computer has Bluetooth 4.0.

That is the most likely cause, alongside general Linux BT stack inefficiencies. In theory people with more in-depth knowledge of the Linux BT stack and/or the companion software might be able to improve it. Otherwise it is how it is.

closed 2022-11-21 by Avamander

Anonymous941, 2022-11-22:
> Very unrelated.
>
>
> That is the most likely cause, alongside general Linux BT stack inefficiencies. In theory people with more in-depth knowledge of the Linux BT stack and/or the companion software [...]

@Avamander Bluetooth 5.0 works, right?  If so I'll buy a dongle.

Avamander, 2022-11-22:
@Anonymous941 
> Bluetooth 5.0 works, right?

That's a bit too generic to base a decision upon. In theory, yes. New BT Low Energy support should be sufficient and newer BT dongles are said to work better, but it's strongly recommended to have the option to return it, should it actually not perform that well.

JF002, 2022-11-25:
I also noticed great variations between devices : my old phone and my PC upload at 1-3KB/s, my new Android phone reach 12BK/s. 
To be honest, I don't know why there are such differences between devices and if there's anything we can do in InfiniTime to improve the upload speed.

Anonymous941, 2022-11-27:
> I also noticed great variations between devices : my old phone and my PC upload at 1-3KB/s, my new Android phone reach 12BK/s. To be honest, I don't know why there are such differences between devices and if there's anything we can do in InfiniTime to improve the upload speed.

@JF002 I've noticed that it occasionally goes more quickly (I'm just guessing here, but I'd say around 5 KB/s) but often goes at around 0.5 KB/s on my computer.  Usually when I first start uploading firmware after a few days of not flashing anything is when it goes fastest.  I have no idea what would cause that strange behavior and am not good enough at C++ to understand driver code...

> my new Android phone reach 12BK/s

Is this a typo or did you mean something different then kilobytes per second?

JF002, 2022-11-27:
Yeah, sorry, that was a typo, I meant KB/s.

Regarding the issue with the upload speed, maybe try with another bluetooth adapter? Or, if you are developing on the PineTime, using the devkit (or an unsealed device) with an SWD probe will definitely be much faster!