InfiniTimeOrg/InfiniTime #1449: Create a new app that receives data from an android device

RazRosman, 2022-11-22:
Hi,

Is there a guide that walkthrough how to implement the ble service in a new app?

I used this guide 
https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/code/Apps.md

to create a new app and it's working great.

Now, I need to use the ble service inside that app to get an integer from an android device. 

I've been reading the code of apps that use the ble, but didn't understand how to implement it my self.

Is there something like that?

Thanks,
Raz

JF002, 2022-11-25:
I don't think anyone has already written such a guide. However, you could have a look at the code of other services for inspiration. For example, here are the PRs that added the weather service : #353 and #756. The services are implemented as new classes in `src/components/ble`.

RazRosman, 2022-12-02:
Thank you very much. The thing is that I don't understand the code.
Can you point me to a developer that I could pay, so he will build it for me?