InfiniTimeOrg/InfiniTime #1453: SMS, EMAIL, etc. empty notification on watch

cr9christi9n, 2022-11-23:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

No contents in notification

### What should happen instead?

Getting begenning of SMS and EMAIL in notification on watch

### Reproduction steps

Hi there,
Watch is able to receive up to 5 message from phone ... begenning of contents in SMS and EMAIL.
I receive a vibration and also a note on the watch but it's EMPTY, there's nothing in it ...

### Version

1.11

### Companion app

Gadgetbrodge v 0.71.3

Avamander, 2022-11-23:
This sounds like you have enabled some kind of privacy setting that stops Gadgetbridge from reading the notification's content. Any chance your SMS and email app are installed in a work profile? Are you using stock email and SMS applications or third-party ones?

In any case, this is not controllable by InfiniTime and should be reported to Gadgetbridge. Though it's worth mentioning that Gadgetbridge's code path for reading and dispatching notifications to companion devices is really really well-trodden. 

It's likely the issue is in the previously mentioned privacy settings (OEM/vendor Android mods), work profile or the apps in question.

closed 2022-11-23 by Avamander

cr9christi9n, 2022-11-23:
I delete every cache files and uninstal gadgetbridge but also last version of Infinitime on my watch to install again.... everything is going on OK now !
Many thanks anyway.