InfiniTimeOrg/InfiniTime #1457: Infineat watch face is missing heart rate monitor icon/reading.

apoorv569, 2022-11-25:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The heart rate icon/reading is missing.

### What should happen instead?

Heart rate icon/reading should appear as in other watch faces.

### Reproduction steps

Change the watch face to Infineat.

### Version

v1.11

### Companion app

Gadgetbridge

JF002, 2022-11-25:
This could be the result of an artistic choice by the author of the watchface. @dmlls, what do you think?

dmlls, 2022-11-26:
@apoorv569 @JF002 Hi, it was indeed a design choice, briefly discussed here: https://github.com/dmlls/InfiniTime/pull/3#issuecomment-1072860833.

In any case, if users think this is a must, I'm open to reconsidering :)

apoorv569, 2022-11-27:
> @apoorv569 @JF002 Hi, it was indeed a design choice, briefly discussed here: [dmlls#3 (comment)](https://github.com/dmlls/InfiniTime/pull/3#issuecomment-1072860833).
> 
> In any case, if users think this is a must, I'm open to reconsidering :)

I see. Well we can definitely wait for some feedback from other users.

InFerYes, 2022-11-30:
I find the HR monitor to be more useful if it can be read from the watch face.

JF002, 2022-11-30:
Thanks for your feedback @dmlls ! 

@apoorv569 If you don't mind, I'll move this post to a [discussion](https://github.com/InfiniTimeOrg/InfiniTime/discussions) to let everyone interested can give their opinion.