InfiniTimeOrg/InfiniTime #1467: Double tapping PineTimeStyle steps style button sends watch to sleep

minacode, 2022-11-30:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

When I tap the buttons of the watch face settings to fast it sends the watch to sleep.

### What should happen instead?

The watch should stay awake and handle two button presses.

### Reproduction steps

1. Set watch face to PineTimeStyle.
2. Long press watch face for settings.
3. Choose lower button (gear icon).
4. See "x" button and "Steps style" button.
5. Double tap "Steps style" button.
6. Watch goes to sleep.

### More details?

This does not happen with the color settings or the settings of the Infineat watch face.

### Version

1.11

kieranc, 2023-03-02:
You're absolutely right. I'll look into it!

kieranc, 2023-03-02:
The issue is line 373 of WatchFacePineTimeStyle.cpp

`if ((event == Pinetime::Applications::TouchEvents::DoubleTap) && (lv_obj_get_hidden(btnRandom) == false)) {`

It disables DoubleTap if the Rnd button is visible, hence the behaviour not happening on the color selection screen. An easy fix would be to check btnClose instead of btnRandom, as it's visible on both color and settings screens. I'll try test/PR it later but if you want to do so, feel free!

PR #1669 (2023-03-04): Fix double tap behaviour on PineTimeStyle settings screen
Fixes #1467 "Double tapping PineTimeStyle steps style button sends watch to sleep"
Double tap is disabled on the color settings screen by checking if the Rnd button is visible, but this didn't work for the options screen as the Rnd button isn't visible. I've changed it to check if the Close button is visible instead, which is used on both settings screens, and resolves the bug. I also changed the button used to disable the long press behaviour which was an as-yet-undiscovered bug which would have allowed the long press action to be used [... cut]
commit ada18233 (2023-03-04): PTS: Fix double tap behaviour on settings screen (#1669)

Fixes #1467 "Double tapping PineTimeStyle steps style button sends watch to sleep"
Double tap is disabled on the color settings screen by checking if the Rnd button is visible, but this didn't work for the options screen as the Rnd button isn't visible. I've changed it to check if the Close button is visible instead, which is used on both settings screens, and resolves the bug. I also changed the button used to disable the long press behaviour which was an as-yet-undiscovered bug which would have allowed the long press action to be used [... cut]
closed 2023-03-04 by Riksu9000