InfiniTimeOrg/InfiniTime #1478: Persistent step storage between reboots

slm85kbh, 2022-12-07:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Storage of logged steps between reboots of the watch

### Description

It would be nice if the logged steps are stored in NVM, because if the watch is rebooted without syncing first, the logged steps are lost.

I don't know if it is a bad idea for the lifetime of the flash, but it might be implemented so that the logged steps are written every 5-10-15 minutes for example, or simple wear-leveling could be implemented.

JF002, 2022-12-07:
I think we have 2 options here :
 - Store the steps in NVM. Since we are using LittleFS, we do not have to worry about wear leveling, as LFS takes care of it. Not sure how it'll impact the lifetime of the flash chip, though.
 - At boot time, do not reset the motion sensor if it's already initialized. This way, the step count won't be reset in the chip and InfiniTime won't have to store the value in NVM.

slm85kbh, 2022-12-08:
I see, I would clearly go for option 2, as a watch reset rarely involves a disconnection of the power supply in a sealed watch. Except when it goes completely flat, but I don't think there's need to account for that. 
It could be mitigated by only storing the step data when the battery is under 10% for example (when the new curve is implemented).
And this method also doesn't put any strain on the flash. 
I haven't looked into the code yet, and I'm fairly new to GIT, so I might not be able to do a PR yet :)

HumanG33k, 2022-12-31:
I think it can permit to sync less the watch with companion app. 
sync only 1 time a week look good  for me as user.

Riksu9000, 2023-01-16:
Duplicate of #493

closed 2023-01-16 by Riksu9000