InfiniTimeOrg/InfiniTime #1496: Nimble central role support

Johnmcenroyy, 2022-12-28:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Pinetime support of external bluetooth sensors 

### Description

Hello
Thank you very much for such a great software.
It will be great if it possible to add Nimble central role to connect to external sensors
such as Polar H10, Polar Stride Sensor, Zwift RunPod, external bluetooth gps (android can be as external gps
device over bluetooth - Bluetooth GPS Output, Share GPS) and so on.

Thanks

Johnmcenroyy, 2022-12-29:
Just investigated a little more.

There is Adafruit Feather nRF52 Bluefruit: 
  https://www.adafruit.com/product/3406
And Adafruit Bluefruit52Lib for working with this module with Central mode example:
  https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Central
Also example of using Bluefruit52Lib for receive data from Polar heart rate sensor (H7,H10):
  https://github.com/MakerLabCRI/PolarBleArduino/blob/master/nRF/nRFPolarH10/nRFPolarH10.ino

Also there is Nordic example for using BLE Central - BLE Heart Rate Collector Example
  https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.2.0%2Fble_sdk_app_nus_c.html
Main source file for this example from SDK (<InstallFolder>\examples\ble_central\ble_app_hrs_c)
  https://termbin.com/fcfq

Maybe it can be integrated in InfiniTime firmware.

Johnmcenroyy, 2022-12-29:
Nim

Riksu9000, 2022-12-30:
Can you explain what the purpose of connecting these devices to InfiniTime is?

Johnmcenroyy, 2022-12-30:
> Can you explain what the purpose of connecting these devices to InfiniTime is?
Hi **Riksu9000**
I have just listed some sensors as example, the matter is to have BLE Central possibility to connect to external sensors.
It can be not just sensors but some bluetooth toys as PowerUP Smartphone Controlled Paper Airplanes and so on.
What concerns certain sensors, I am espacially interested to connect from Pinetime to Polar H7 to get medical grade heart rate data, to get R-R intervals data to analyze stress, also there is a possibility to get aerobic threashold, anaerbic threashold with DFA Alpha 1 from HRV from chest heart rate.

So the main point is the possibility to connect to external sensors using NimBLE Central possibility and let the user decide which sensor and how to use it, let him write some additional apps for certain sensors.

Peter Gamma wrote also about this here:
https://forum.pine64.org/showthread.php?tid=11308

Johnmcenroyy, 2022-12-30:
Polar H7/H10 Chest Heart Rate Sensor
  medical grade heart rate data
  hrv data (r-r intervals)  
    hrv give the possibility to get stress index, aerobic threashold,
    anaerobic threashold, heart rate problems etc.

Polar Stride Footpod
   speed
   distance
   leg cadence
   stride length

Zwift Runpod Footpod
   speed
   distance
   leg cadence

Footpods when calibrated can be accurate as gps and more.
Work in forest, indoor etc.

PowerUP Smartphone Controlled Paper Airplanes
   just toys