InfiniTimeOrg/InfiniTime #1500: Motion sensor doesn't work anymore

realpixelcode, 2022-12-29:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The PineTime does not sense any motions anymore.

### What should happen instead?

It should sense motions, count foot steps, wake on raise etc.

### Reproduction steps

I don't know, it just stopped working from one day to another. Perhaps the hardware is broken? But I couldn't imagine why it would be – the watch didn't fall onto the ground or anything.

### More details?

The step counter, Shake Wake, Raise to Wake and XYZ motion diagram don't work at all anymore.

### Version

v1.11.0

minacode, 2022-12-29:
> Perhaps the hardware is broken?

This is very likely and one of the more frequently reported problems.

realpixelcode, 2022-12-30:
Hm, is there any way of checking whether it's a software or a hardware problem? (And if it's a hardware problem, can it be repaired?)

minacode, 2022-12-30:
You could try different softwares. If the problem persists it is most likely a hardware problem. 

I don't know about repairing. I guess its possible in theory.

realpixelcode, 2022-12-30:
Different software? I don't know of any alternatives to InfiniTime, unfortunately 😕

minacode, 2022-12-30:
You can try older versions of InfiniTime or WaspOS. But I would bet on a hardware problem.

Gernomaly, 2023-01-03:
I had the same problem. It was sadly hardware related. I am currently still thinking of a good way to fix it, without having to buy a new one

realpixelcode, 2023-01-03:
Okay, will close 😕

Mindavi, 2023-01-04:
You can see if the sensor is detected in the about menu. If there's question marks it's very likely a hardware issue

realpixelcode, 2023-01-04:
It says:

> Accel. ???

So it's indeed a hardware defect. 😕 I guess it's time for a Bangle.js. Thanks for your help :)

closed 2023-01-04 by realpixelcode

sammyj85, 2024-02-17:
I'm getting the ??? for Accel also.... sad.

Reflashing the firmware didn't help.