InfiniTimeOrg/InfiniTime #1517: Wakeup no longer working (tapping, shaking, turn wrist)

ReaderGuy42, 2023-01-07:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The enabled wakeup options no longer work

### What should happen instead?

I was expecting the watch to "wake" when tapped or shaken

### Reproduction steps

1. ??
2. tap watch
3. not waking up

### More details?

I've reinstalled the firmware (1.11) and it still doesn't wake. I've played around with the one tap, two tap options, but nothing works to wake the watch, besides pressing the physical button.

I also noticed that the shake calibration is completely unresponsive.

I've only had the device for a few days and originally this all worked. I think (!) when I first installed the updated 1.11 it also all worked, but as of yesterday it no longer does.

### Version

1.11

### Companion app

GadgetBridge

JF002, 2023-01-07:
The sleep mode is probably enabled : swipe right to open the quick actions menu and check the bottom left icon. If it's purple, it means that the sleep mode is enabled, which disables all wake options. 
[image]

Tap on it until it turns green to re-enable them.

ReaderGuy42, 2023-01-07:
Yup, that was it, that was easy. Thanks!!

closed 2023-01-07 by ReaderGuy42