InfiniTimeOrg/InfiniTime #1522: Disable notifications pop-ups while using torch app

rambonette, 2023-01-12:
First of all I want to point out that I use the torch app quite a lot and I find it very useful.
Allow me to take a few moments to explain the situation to you:
Last night I was using the torch app in the dark to find my storage's lock keyhole. All of the sudden a message arrrives on my phone which rightly triggers a notification on the watch. Not only the torch was momentarily deactivated, but once the notification timer expired it returned to the watch face screen. As you can imagine it was quite inconvenient having to re-open the torch while having a bunch of keys in my hand.

Shouldn't we make sure that when the torch is active, notifications are blocked? Or that if notifications come in, when they expire, we revert to the torch app?

Steps to reproduce issue:
- Open torch app and turn on torch
- Trigger a test notification from a companion app (like Gadgetbridge)

I really don't know if this we should label this as bug or enhancement.

I'd tackle this myself, but I'm sadly overworked right now. So I want to thank in advance anyone who wants to address this issue!

FintasticMan, 2023-01-12:
It would definitely be possible to prevent notifications while the torch is on. The other issue of it not returning to the torch is actually fixed in the latest development branch (PR #1350), so it will be available in the next release.

rambonette, 2023-01-12:
Sorry

Riksu9000, 2023-01-30:
Duplicate of #610