InfiniTimeOrg/InfiniTime #1534: Store shake/wake sensitivity settings to external memory and reload it on boot-up

Danimations, 2023-01-16:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

It would excellent if the Pinetime watch would recover its shake/wake sensitivity after the battery has run flat. Presently, this is lost when the battery runs flat. After a flat battery, the shake/wake setting defaults to a setting which is very conservative, meaning that waking the screen typically requires the press of the device's button, and therefore two hands to perform.

### Preferred solution

I gather that the setting data could be stored on the device's external memory, and reloaded on boot-up. This would be helpful for those of us users who occasionally let the battery run entirely flat before recharging it, and who like to wake the device with a gentle flick of the wrist.

### Version

1.11.0

FintasticMan, 2023-01-16:
The shake wake threshold is stored in the settings controller, which means that it should be stored on the external flash along with all the other settings.

minacode, 2023-01-16:
The external memory doesn't work without battery. The settings are already stored on the external storage as a file, but once the battery runs out, everything is lost.
I guess the best solution would be some kind of settings backup via a companion app.

FintasticMan, 2023-01-16:
I believe that the external memory is SPI NOR flash storage, which is able to keep data without any power AFAIK.

minacode, 2023-01-16:
Oh, ok. I didn't know that.

Riksu9000, 2023-02-22:
The shake threshold is saved like other settings. Can't repro.

closed 2023-02-22 by Riksu9000