InfiniTimeOrg/InfiniTime #1552: Enhancement: title ticker scroll starting point/timing adjustment on notifications

davidleebooth, 2023-01-21:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

Hello, I have noticed when swiping to the notification tray and when the notification pops up, the beginning of the notification titles have already scrolled by by the time my eyes meet the notification, resulting in waiting for the beginning of the title to scroll by again. 

### Preferred solution

When swiping into the notification tray or when a new notification pops up, the title resets its scroll with beginning in the center of the screen, giving users enough time to start reading along before the beginning disappears, or a moments pause before the scroll begins. Another idea would be long press the notification to restart the title scroll

### Version

1.11.0

SteveAmor, 2023-01-21:
I too suffer from this issue. I think a moments pause (2 secs) would work for me.

FintasticMan, 2023-01-21:
One implementation of this is #1438.

Riksu9000, 2023-01-21:
LVGL 8 has a built in way to delay the scrolling. #616

Riksu9000, 2023-02-21:
It looks like the scrolling starts while the screen is still transitioning. We should look at fixing that before adding delay. Screen switching in LVGL should be done with lv_scr_load(_anim), but we don't use that, which may be causing this issue.