InfiniTimeOrg/InfiniTime #1589: Infinitime does not compile in Debug mode

BloodStainedCrow, 2023-02-16:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

When compiling InfiniTime with the buildType Debug using a devcontainer the compile fails because of the resulting size overflowing the available space.

```
[build] /opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: pinetime-app-1.11.0.out section `.text' will not fit in region `FLASH'
[build] /opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: region FLASH overflowed with .data and user data
[build] /opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 30128 bytes
[build] Memory region         Used Size  Region Size  %age Used
[build]            FLASH:      504760 B     474632 B    106.35%
[build]      SPARE_SPACE:          0 GB        12 KB      0.00%
[build]              RAM:       56008 B        64 KB     85.46%
[build] collect2: error: ld returned 1 exit status
```

### Preferred solution

Ideally it should be possible to compile in Debug mode obviously, but if that is not possible removing it as an option (or at least not making it the default option) might prevent new contributers from being confused that the default compile fails.

### Version

develop

Avamander, 2023-02-16:
The default build type is `Release`. The release modes, `Debug` and `Release`, are documented in `docs/buildAndProgram.md`, including the part about not fitting in flash.

As there is no trivial way to make `Debug` builds fit the flash while also keeping all the parts that are relevant in that specific case, it's up to the person wanting to debug to choose what to leave out and to fine-tune the build to fit their needs.

closed 2023-02-16 by Avamander

JF002, 2023-02-20:
@BloodStainedCrow If you need more help about this, please check our [Discussions board](https://github.com/InfiniTimeOrg/InfiniTime/discussions), as this question may have already been asked. Feel free to open a new post if you can't find it.