InfiniTimeOrg/InfiniTime #1592: The current time cannot be seen in apps

Riksu9000, 2023-02-19:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

The time is only displayed in quicksettings, applist and the watchface. The time cannot be seen when using an app. We also don't have the screen space to spare for displaying the time in every app.

### Preferred solution

Instead of putting the time in apps, the time could be temporarily overlaid on the screen, in a small font at the top edge of the screen for example, every time the watch is woken up, and perhaps with a strong shake as well.

minacode, 2023-03-04:
I would like to have a lock-screen that appears first on wake-up before the watch goes to the last open screen. It would make sense that it would be (similar to) the watch face. 

All we need to show the time with an "overlay" would be a method to reach the lock-screen from the running system (without the lock of course). I am thinking of a swipe gesture or long touch or something like that.

My phone works like that.

Riksu9000, 2023-03-04:
This would be a problem if you have a timer for example that you keep open so you can quickly check it. It would be the same for stopwatch, if it didn't disable sleep.

minacode, 2023-03-04:
That's true. Keeping them running in the background and showing them on the lock screen if needed is the common solution for this, I guess.

JF002, 2023-03-27:
> Instead of putting the time in apps, the time could be temporarily overlaid on the screen, in a small font at the top edge of the screen for example

Maybe we can use the top and sys layers from LVGL for this?

https://docs.lvgl.io/7.11/overview/layer.html#top-and-sys-layers

lorhof1, 2023-04-14:
It is already possible to go to notifications by double-pressing the button and then go back to the previous app with a single press or upwards swipe. The easiest solution would probably be to add the time to the notifications screen as it can be opened from apps already.