InfiniTimeOrg/InfiniTime #1644: Event messaging improvement idea with std::function

Riksu9000, 2023-02-25:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

With the PushMessage functions, there's a lot of opportunities for errors. Anything can push anything through that function.

### Preferred solution

I had the idea that we could use std::function to pass an event callback function to classes. This way classes can be made much more generic without SystemTask dependencies, and safer.

https://github.com/Riksu9000/InfiniTime/commit/e67451d15e10716bf9c0a4e30edd86d755068430 Here's how this could be used to make TimerController much more generic. The issue with this is that it seems to use more memory, and I'm not sure why that is. If you can, please review this idea and let me know what you think. Is this a good approach? Can it be improved?

JF002, 2023-02-26:
I don't know all the internals of std::function, but, from what I know, `std::function` and lambda can be very optimized (for example, in some conditions like when the capture is empty, they are optimized as a simple function pointer) or very heavy (with dynamic allocation and a lot of processing).

In this case, I guess the increased memory usage comes from the capture in the lambda.

Riksu9000, 2023-02-27:
I was able to reduce the memory usage of this a bit. https://github.com/Riksu9000/InfiniTime/commit/469ce4d2e639f9eedc79cca97f3d2cdbaceb7eae