InfiniTimeOrg/InfiniTime #1673: Infinitime will not boot after the battery fully dies

adryd325, 2023-03-06:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I left my watch off for a week and it won't boot

### What should happen instead?

It should boot

### Reproduction steps

Leave the watch's battery to die

Attempt to power on after a long time

### More details?

I tried holding the side button to reboot it, holding it until the pine symbol turned blue and held it till it turned red and none of these worked

### Version

v1.11.0

ajack2001my, 2023-03-06:
Leave it on the charger for a few day, it will eventually start-up, then keeps rebooting until there is enough charge in battery to allow a "normal" boot. That will then let it charge till full.

adryd325, 2023-03-06:
it's stuck at the infinitime slash screen for hours