InfiniTimeOrg/InfiniTime #1676: Motion sensor seems broken

the-voidl, 2023-03-07:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

PineTime displayed insanely high number of steps for a few days. Now it does not count any more.

### Reproduction steps

Two weeks ago my PineTime started displaying a step count of 536,876,736 out of nothing. I can't exactly tell when it begun.
I thought it was a small bug, as it could happen in FOSS. But I was wondering why wake up by wrist raise didn't work any more.

### More details?

The high step count persisted during reboots and days without wearing the watch.

Now I played around a bit more and first tried to calibrate the sensor and afterwards opened the XYZ-diagram app. Both of the functionalities don't work for my device. Step count stays at 0 and xyz-diagram doesn't show anything than a straight line.
The previously used version was 1.11, after upgrading to FOSDEM Version I couldn't notice any difference.

### Version

v1.11.0

### Companion app

Gadgedbridge

the-voidl, 2023-03-07:
Just found #1446 and #1500

My about menu also shows "Accel. ???" so it is a hardware defect.

Could this be triggered by v1.11? I personally don't think so since I flashed 1.11 in November and this issue started in mid February...

closed 2023-03-07 by the-voidl

JF002, 2023-03-26:
> Could this be triggered by v1.11? 

This is most probably a hardware issue. InfiniTime displays those `???` when it cannot communicte with the motion sensor.