InfiniTimeOrg/InfiniTime #1679: Purpose of MotionController

Riksu9000, 2023-03-09:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

What is the purpose of MotionController? We just copy data from the driver to it and add a few functions. Why not use the driver directly?

### Preferred solution

Remove MotionController and use the driver directly. The trip counter could be another class, maybe the step counter as well.

JF002, 2023-03-20:
The idea behind `MotionController` (like many other `Controller`s like `MotorController`, `BatteryController`, `BrightnessController`,...) is to decouple the application from the low-level driver and the hardware.

The goal is to implement only the low-level processing in the driver : configure the sensor, read raw data from it, maybe handle IRQ, DMA,...
Then, the `Controller` layer uses this low-level driver to provide a higher level service and API to the application. 

Some of the reasons for this design are
 - The low level drivers are simpler : they only expose the data from the sensor, with no additional processing or logic
 - The application does not know anything about the hardware. You can switch to another driver (that exposes the same API) without changing anything in the application. This is quite useful to support other sensors or write unit tests.
 - All the additional logic and data processing specific to InfiniTime is implemented in the components.

This is probably not clear if you only look at `MotionController`, but look at the `HeartRateController`. The application cannot directly use the raw data from the sensor. So, instead of converting data in the low level driver of in the GUI, the processing is implemented in the `Controller` layer.

`MotionController` adds the ShakeWake and RaiseWake algorithm, so I wouldn't say it's useless.

Also, removing the `MotionController` and directly using the low level driver breaks the intend of the design (application does not directly talk to the hardware) and is not consistent with other parts of the projects.

Riksu9000, 2023-03-20:
Sorry I wasn't clear enough in the main post. I mainly wanted to bring attention to the fact that the controller was used to access raw sensor data, and hadn't planned out what should be done about it entirely. The PR #1700 should address this appropriately.