InfiniTimeOrg/InfiniTime #1685: Metronome: Dragging the arc will snap to 98 bpm

Riksu9000, 2023-03-12:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Dragging the arc will snap to 98 bpm

### Reproduction steps

This issue happens since this commit https://github.com/InfiniTimeOrg/InfiniTime/commit/5ef3c0d3be58b6741a9e443e6d2151de2b3a59bc.

### More details?

In Metronome, we don't want the app to close while dragging the arc. This was previously accomplished by setting the return touch event to none, and handling the exiting in app, but there's currently no way to disable the default return touch event. Returning true in OnTouchEvent will cancel LVGL tap, which will move the cursor to {-1, -1}, which is at the top left corner of the screen, setting the bpm to 98. Returning false will allow DisplayApp to close the app if it detects an appropriate touch event.

### Version

develop

minacode, 2023-03-12:
What do you think about using the number choosing widget like in the time related apps? The arc looks good, but it is not very practical in general.

Riksu9000, 2023-03-12:
If there's a proposal that improves upon the current state of the app, we'll consider it, however this issue limits how we can use LVGL in general and should be fixed regardless.