InfiniTimeOrg/InfiniTime #1690: Switch to another docker container registry

JF002, 2023-03-15:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

I received an email from Docker yesterday. It's titled : "Docker is sunsetting Free Team organizations".

Basically, they decided to stop offering the "Free team organization" account I registered for the [InfiniTime account](https://hub.docker.com/orgs/infinitime/members) on Docker Hub. 

They kindly ask me to upgrade to a paid (300$/yr) subscription to keep the account, or to loose the account and all the data in 30 days (April 14, 2023).

This issue is widely discussed here : https://github.com/docker/hub-feedback/issues/2314

InfiniTime uses Docker Hub to provide the build container to the users and the to Github Actions workflows. It basically consists in a single image built for x86_64 and AARCH64. 

I would like to take this opportunity to look for a better alternative than paying or applying for their open source program (do we really trust them to keep their engagement at this point?)

### Preferred solution

I think we have a few free and FOSS friendly alternatives:

- [Codeberg.org](https://codeberg.org/) might offer a container registry
- I can allocate a resources on a VPS to selfhost the registry
- GitHub Container Registry might simply do the job and is free for public repositories like InfiniTime

Since InfiniTime is hosted on Github, using GHCR (GitHub Container Registry) is probably the simplest and more efficient solution, assuming github won't change their mind about the pricing.

What do you think about this?
Also, what would it take to switch our github Action workflows to another registry?

FintasticMan, 2023-03-15:
It seems to me that using GHCR (or, if that also works, Codeberg) would definitely be a good enough solution. So I don't think it's necessary to set up a VPS to host it, that would just cause headaches IMO.

Switching the workflows to push to a different registry should be really simple, just prefixing the image names with the registry address and updating the secrets.

If we decide on GHCR, it should also be possible to upload images from pull requests, so that they can be tested before merging more easily.

Vad1mo, 2023-03-15:
If, for some reason, the Docker-Sponsored Open-Source Program isn't working out, we are happy to help you out. We operate a Harbor-based [container registry as a service ](https://container-registry.com/)that has many benefits over most of the other registries out there.

There are also features regarding containerized image distribution, that might be valuable too as well.

Don't underestimate the value of owning and controlling your data, it is easier to migrate if the domain is under your control.

fossison, 2023-03-16:
Re: @Vad1mo 's mention of Docker-Sponsored OSS, @JF002 you may be able to apply for it here: https://www.docker.com/community/open-source/application/

I'm not sure if they would then migrate the existing account or require it to be re-made, but it may be the path of least resistance.

JF002, 2023-03-26:
Ok, Docker announced that they finally won't delete the free team accounts : https://www.docker.com/developers/free-team-faq/?utm_campaign=2023-03-24-ft-sunset-reversal-ft&utm_medium=email&utm_source=marketo&mkt_tok=NzkwLVNTQi0zNzUAAAGKtH5al1IwI58BEMHfcqR1E0D-8VdY_uoFqd70DPLV3eiZM1uxLVNR8DGqWBdBZX12G-FbsznAFcKT4wY_FEAlhZ35cK8ZqgfsoC2eR17OOBk

So there's no need to change our docker repository right now. I'll also apply to they open source program in case they eventually decide to change their mind once again. I'll keep this issue open to post an update when I'll receive an answer from Docker.

JF002, 2023-05-07:
InfiniTime is now a *community organization* on Docker Hub. The subscription is valid until May 2, 2024. We'll see what happens at that time :)

closed 2023-05-07 by JF002