InfiniTimeOrg/InfiniTime #1701: Infineat face, AM/PM indicator not working

Rathmox, 2023-03-18:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

an M is showing up in the top right corner

### What should happen instead?

AM or PM sould be showing up in the top right cornet

### Reproduction steps

1. Start the watch
2. Set the watch face to "Infineat face"
3. Set the time format to 12-hour
4. on the time screen, an M is displaying in the top right corner

### More details?

picture found on Reddit
[image]

### Version

v1.11.0 FOSDEM Edition

### Companion app

Gadgetbridge

SteveAmor, 2023-03-18:
Same on standard V1.11.0

SteveAmor, 2023-03-18:
I'm assuming https://github.com/InfiniTimeOrg/InfiniTime/blob/310ea81eec3bd35e13718dd85e578048ce20eb47/src/displayapp/screens/WatchFaceInfineat.cpp#L443 needs to be ```lv_label_set_text_fmt(labelTimeAmPm, "%s", ampmChar);``` but my dev setup is not in a state to test.

SteveAmor, 2023-03-18:
> I'm assuming
> 
> https://github.com/InfiniTimeOrg/InfiniTime/blob/310ea81eec3bd35e13718dd85e578048ce20eb47/src/displayapp/screens/WatchFaceInfineat.cpp#L443
> 
> needs to be `lv_label_set_text_fmt(labelTimeAmPm, "%s", ampmChar);` but my dev setup is not in a state to test.

This didn't fix the problem

Riksu9000, 2023-03-18:
https://github.com/InfiniTimeOrg/InfiniTime/commit/5feccb382a86d304e7cfcdb5a5b04c109bf7771f This will be fixed in the next release.

closed 2023-03-18 by Riksu9000

ScanuNicco, 2023-05-02:
I recently upgraded to v1.12.0 but I'm still experiencing this issue:

[image]
[image]

Riksu9000, 2023-05-09:
You need to update the resources.

ScanuNicco, 2023-05-09:
Alright, i'll give that a shot. If this is the case, it might be worth updating the [1.12 release notes](https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.12.0) as they say
> The resources package was not updated for this release, so there's no need to re-install it.

JF002, 2023-05-10:
@ScanuNicco Oh sorry, my bad, I missed [that little change in the resource file](https://github.com/InfiniTimeOrg/InfiniTime/commit/5feccb382a86d304e7cfcdb5a5b04c109bf7771f) when I wrote the release note. I've just updated it. 

Sorry for the confusion!