InfiniTimeOrg/InfiniTime #1711: Cannot compile because of maxLapCount

targor, 2023-03-28:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The project  will not compile completely due to a maybe declaration error

### What should happen instead?

it should compile without errors

### Reproduction steps

used the tutorial for building the poject: https://github.com/InfiniTimeOrg/InfiniTime/blob/main/doc/buildAndProgram.md

downloaded everything, installed every package and tried to build with the command in the tutorial:
make -j4 pinetime-app

### More details?

This bug will hapenn if i try to compile:

`/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: region FLASH overflowed with .data and user data
/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 18428 bytes
/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/pinetime-app.dir/displayapp/screens/StopWatch.cpp.o: in function `Pinetime::Applications::Screens::StopWatch::stopLapBtnEventHandler()':
/InfiniTime/src/displayapp/screens/StopWatch.cpp:192: undefined reference to `Pinetime::Applications::Screens::StopWatch::maxLapCount'
Memory region         Used Size  Region Size  %age Used
           FLASH:      493060 B     474632 B    103.88%
     SPARE_SPACE:          0 GB        12 KB      0.00%
             RAM:       56000 B        64 KB     85.45%
collect2: error: ld returned 1 exit status
make[3]: *** [src/CMakeFiles/pinetime-app.dir/build.make:1819: src/pinetime-app-1.11.0.out] Error 1
make[2]: *** [CMakeFiles/Makefile2:423: src/CMakeFiles/pinetime-app.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:430: src/CMakeFiles/pinetime-app.dir/rule] Error 2
make: *** [Makefile:241: pinetime-app] Error 2`

### Version

main

### Companion app

Usually Gadgetbridge, but in this case it does not matter.

Avamander, 2023-03-28:
If you want to create a debug build you have to be more selective about what parts you compile with what flags. It won't all fit.

closed 2023-03-28 by Avamander

targor, 2023-03-28:
How do i do that?

Avamander, 2023-03-28:
Modify the CMakeLists.txt and remove/add `-gN` where `N` is the desired debug information amount.

Generally though, just use the release build and enable debug info for just the components you want.