InfiniTimeOrg/InfiniTime #1724: Time sync broken in 1.12

omartijn, 2023-04-02:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Time/date stopped syncing in 1.12

### What should happen instead?

Time and date should sync to the phone

### Reproduction steps

Update your PineTime to 1.12, connect it to AmazFish on SFOS. The time/date is stuck on Jan 1 1970. It should automatically sync on connection. Other things do work fine (e.g. notifications).

### Version

1.12

### Companion app

AmazFish 2.1.2-1.5.2

Avamander, 2023-04-02:
Have you tried a different companion app?

Shatur, 2023-04-02:
Can confirm with watchmate compiled from source right after the release.

Narann, 2023-04-02:
Same for me with Gadget Bridge: https://www.reddit.com/r/pinetime/comments/129lb7v/infinitime_112_for_pinetime_released/

I will wait before validating the firmware.

Thanks to the contributors, keep the good job!

Shatur, 2023-04-02:
Works with watchmate for me.

Self-Perfection, 2023-04-02:
Gadget Bridge also hangs on "connecting" for me. So I have rebooted PineTime to previous firmware and now connection works fine.

ThomasHertleif, 2023-04-02:
I also had Gadget Bridge getting stuck on "connecting".  
Deleting and pairing the watch again fixed the issue for me.
The time was also set correctly.

Avamander, 2023-04-02:
Testing conducted in the PineTime channel determined that Amazfish is most likely writing a byte or two too few to the Current Time Service and thus they get ignored. The service was updated to be more standards-compliant.

tomechio, 2023-04-02:
I experienced similar problem few weeks ago when I built the update file by.myself:
https://github.com/InfiniTimeOrg/InfiniTime/discussions/1702

ricsim, 2023-04-02:
Same here.
I am using GadgetBridge on Android.
I do not have the ressource pack installed, as GB does not yet have support to flash that.
When I update to 1.12 it does not connect via BT, and hence the time is 00:00.
The only way to get it to connect is to hold the button until it goes back to 1.11 which works. (Was also installed with GB. I tried to install 1.12 3 times with allways the exact same result.)

Narann, 2023-04-02:
I just tested this:

You remove the pairing in Android setting, shutdown Gadget Bridge (Quit). Restart Gadget Bridge and try to connect: It now connect.

So maybe the Android pairing is causing the issue.

Jed28, 2023-04-03:
> You remove the pairing in Android setting, shutdown Gadget Bridge (Quit). Restart Gadget Bridge and try to connect: It now connect.

Confirmed to fix the GB issue. You **must** Quit GB from it's menu not android force stop.

mkincaid, 2023-04-03:
Are we sure this is companion app-specific? I'm finding that setting time on a 1.12 watch is ignored with latest InfiniLink for iOS (1.0.2) as well as HubMartin's [WebBLEWatch](http://hubmartin.github.io/WebBLEWatch/) page. Both worked on the same watch in 1.11. There are no issues with pairing or doing other activities (e.g. WebBLEWatch can read the current time from the watch, just not set it).

Avamander, 2023-04-04:
@mkincaid Yes, companions that aren't writing the correct data to the characteristic can't set time.

JF002, 2023-04-16:
I think https://github.com/InfiniTimeOrg/InfiniTime/commit/38092fcb40695098702163ab64a06787b2dc2499 introduced changes that potentially broke companion apps that implemented the CurrentTime Service/Client according to our previous implementation .

Both CurrentTimeService and CurrentTimeClient check that the received data is at least the size of the time struct. Since we added 1 byte to the structs, companion apps that only send the amount of bytes we previously needed cannot set the time anymore.

I always try to avoid breaking companions apps without notifying the developers but... I failed this time!

So, the first think to do is to notify developers of those companion apps, and even better open a PR with the changes needed. 

We can also optionally implement a simple fix in CurrentTimeService and CurrentTimeClient to accept the older format  so that the changes are transparent for the users. This will also allow InfiniLink to continue working (since it's not maintained anymore).

JF002, 2023-04-16:
I've already created the [PR in Amazfish](https://github.com/piggz/harbour-amazfish/pull/261).

Gadgetbridge 0.73.0, Siglo, Watchmate and ITD work fine on my side.

I cannot test InfiniLink, unfortunately.

Avamander, 2023-04-17:
There's the unfortunate aspect that we documented that the service follows the CTS specification but we used a byte less than we should've. Apps that implemented CTS as specified kept working.

It would have been better to give more of an heads up but nobody predicted some implementations would be non-standard to rather follow InfiniTime than CTS.

Avamander, 2023-04-17:
@dennisguse Remove the watch and re-add it.

Self-Perfection, 2023-04-17:
For me bluetooth in version 1.12 seems broken. I tried deleting InfiniTime from list of connected devices in Android but I can not re-add it. Other devices do not see it on bluetooth scan. I tried toggling bluetooth in InfiniTime settings off and on again but it did not help. So I rolled back to 1.11.

dennisguse, 2023-04-17:
I cannot re-add with 1.12. 
Gadgetbridge does the pairing successfully and then the device lost remains empty. 
Did even clear all data for Gadgetbridge. 
Rolling back to 1.11. works fine again.

Avamander, 2023-04-17:
@Self-Perfection You have to remove the device from both Android settings and Gadgetbridge. Gadgetbridge will not display devices that are already bonded.

In any case, getting rid of any inconsistent state your phone is holding on to will resolve the issue. Maybe you have to reboot your phone in-between, but those problems are not in the scope of this issue.

Wivik, 2023-04-20:
I've had the same Gadgetbridge issue (no time sync, device missing while bonded), solved with the following actions : 

1. Forget the device in Android
2. Clear Gadgetbridge's storage
3. Reallow every Gadgetbridge permissions
4. Bound device
5. Pinetime displayed in Gadgetbridge, time synced

closed 2023-04-30 by Avamander