InfiniTimeOrg/InfiniTime #1736: Substantially reduced battery life after 1.12 upgrade

lightweight, 2023-04-10:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

After upgrading my Pinetime to 1.12 I see substantially reduced battery life

### What should happen instead?

I would expect similar battery life as experienced on earlier versions.

### Reproduction steps

After upgrading my Pinetime from 1.11 to 1.12 (via GadgetBridge) I found that my watch always runs out of charge within 10-12 hours of being put on in the morning, whereas prior to the upgrade I never had the watch battery run down in a single day. 

### More details?

I believe that I've had short battery life every day since the upgrade, and no problems prior. 

I accept that it's possible the battery has deteriorated coincidentally with the upgrade, but given the suddenness, I suspect a change with the upgrade to 1.12. Have any new services been turned on by default? I do not normally use the pulse monitor at all. On one day it started itself while I was walking in the rain,  but I think it only ran on that one day and only for not more than an hour. 

### Version

v1.12

### Companion app

GadgetBridge 0.73.0

RageGamerBoi, 2023-04-15:
That's because of the better battery curve and estimation.
You likely have the same battery life, but before it appeared to have incredible battery life from 100% to 50%, but under about 40% the battery would appear to drain to 0% in a matter of hours.
If it's something that creates a noticeable difference in time from 100% to 0% that's a different story, and I'd recommend downgrading, and timing how long the watch takes to die, then testing on 1.12 to see the difference.
I currently believe, however, that it's just your brain getting a little confused and you'll get used to the better estimation soon.

Riksu9000, 2023-04-16:
I don't think any changes were made that would significantly impact battery life. The way the battery capacity was calculated was changed, but that wouldn't change how long it takes until the watch shuts down.

Can you try charging the watch until it says 100% and see how long it lasts then? It will stay at 99 for a bit until it's actually full. You can also try downgrading the firmware and see if the issue goes away.

JF002, 2023-04-16:
I did not notice any change regarding the battery live on my pinetimes, and as @Riksu9000 said, I can't find any change that could explain this.

Could you maybe check the settings to ensure that wake up options and display timeout are correctly set? Maybe also check that the HR sensor is not running.

Does the watch correctly goes to sleep (the displays shuts down) after time specified timeout?

Also, note that the battery level is probably a bit optimistic when the watch is charging. Have you tried charging the battery for more time (even if it displays 99/100%) ?

closed 2023-04-18 by lightweight

JF002, 2023-05-01:
Thanks for the update, @lightweight ! Happy that you found a solution to this issue!