InfiniTimeOrg/InfiniTime #1746: "Date&Time" to "Date & Time" in Settings

RageGamerBoi, 2023-04-16:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

It's a minor formatting change, but it's relatively annoying.

### Version

v1.12.0

Elements6007, 2023-04-17:
From my personal standpoint "Date&Time" looks better. Adding space makes it look awkward and long. 

[image]

RageGamerBoi, 2023-04-18:
> [image]

That looks better imo...
I think it only looks a little weird because there aren't any other buttons with 2 words on them on that page.
It matches the first settings page better though.

Elements6007, 2023-04-19:
> 
> That looks better imo... I think it only looks a little weird because there aren't any other buttons with 2 words on them on that page. It matches the first settings page better though.

True, I guess it does match better.

ThatOneCalculator, 2023-04-28:
Why not `Date/Time`?