InfiniTimeOrg/InfiniTime #1767: Pinetime unable to charge after battery empty

Hensch-S, 2023-05-31:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

After completely discharging, the Pinetime does not boot anymore.

### What should happen instead?

Pinetime should start up normally.

### Reproduction steps

Let the battery discharge completely until it turns off. Then try to charge. The boot loader does not come up, display is flickering (on/off)) and the battery isn't charged.

### More details?

This is the first complete battery discharge after Firmware 1.12 was flashed. When I plug in the Pinetime to charge the battery, the display begins to flicker and the battery isn't charged. I let it over night in this state and nothing changed. Maybe it is now in a deep discharge state from which it cannot recover by itself.

### Version

1.12

### Companion app

Gadgetbridge

FintasticMan, 2023-05-31:
This isn't an InfiniTime issue, but rather a hardware issue. To get it to charge, try plugging it into a computer's USB port, those seem to work best, and then leaving it on the charging cradle for a long time.

If that doesn't work, some people have reported success by applying 6v to the charging pads on the back of the PineTime until it powers on. After that, you can transfer it over to the charging cradle.

closed 2023-05-31 by FintasticMan

neilmayhew, 2023-06-30:
I had this problem for several months too, but the improved voltage/charge curve in the last firmware update (1.12.0) has improved things and now the watch doesn't seem to get as deeply discharged before shutting down.

Avamander, 2023-06-30:
The displayed curve does not affect the charge controller. Neither have we ever had a "shutdown", for multiple reasons and lack of hardware support being one.

a13xie, 2023-06-30:
Indeed. Linking one of the issues where this was explained for easier searchability.

https://github.com/InfiniTimeOrg/InfiniTime/issues/201

neilmayhew, 2023-08-07:
I guess the reason I'm not getting the problem is that I'm more careful not to let the watch get fully discharged now. The colour change in the battery indicator really helps with that, so perhaps that's the reason why I haven't had the problem since version 1.12.0.

neilmayhew, 2023-08-07:
One trick that seems to help me when the watch gets into this state is to pick up the cradle and watch and squeeze the watch firmly into the cradle, holding it there for 2-3 minutes. I assume this improves the contact between the charging pins and the pads, and increases the voltage just enough to get things going. I think there may be a bigger power draw during the boot sequence, so if I can get it as far as the regular display it's often OK on its own from then on.

Cleaning the pins and the pads (eg with an eraser) also helps to improve the contact.