InfiniTimeOrg/InfiniTime #1778: Pinetime not updating

MrMjau, 2023-06-25:
I have tried to update to the 1.13 release, but it just doesn't seem to work. I have tried updating it using gadgetbridge and nrfconnect but my pinetime keeps saying failed

a13xie, 2023-06-25:
I have updated using Gadgetbridge without any issues.

FintasticMan, 2023-06-25:
Ss it says in the README, nRF Connect isn't supported anymore. Try again with Gadgetbridge, and otherwise I'd recommend trying with a different device.

ricsim, 2023-06-25:
I could update with Gadgetbridge, but Bluetooth does not work.

It does not connect to my Mobile any more.

(Same with 1.12)

So I have to reboot to 1.11 which works again.

 

Any idea how to debug or why it does this with 1.12 und 1.13?

 

CU, Ricsi
 

Gesendet: Sonntag, 25. Juni 2023 um 11:04 Uhr

MrMjau, 2023-06-25:
> As it says in the README, nRF Connect isn't supported anymore. Try again with Gadgetbridge, and otherwise I'd recommend trying with a different device.

I also tried using my iphone and infinilink but it still didn't work

a13xie, 2023-06-25:
Have you tried restarting the watch?

Bugsbane, 2023-06-25:
I just updated from 1.11 to 1.12 last week without any issues, but trying to update to 1.13 today fails the same way as the OP. I'm using GadgetBridge 0.74.0 (with all permissions) on a Samsung A51 with Android 13.

MrMjau, 2023-06-25:
> I just updated from 1.11 to 1.12 last week without any issues, but trying to update to 1.13 today fails the same way as the OP. I'm using GadgetBridge 0.74.0 (with all permissions) on a Samsung A51 with Android 13.

Thats interesting @Bugsbane maybe it's an issue with infinitime 13

hawkinspeter, 2023-06-25:
I'm also struggling to update from 1.12 to 1.13 using GadgetBridge 0.74.0d-banglejs on Android

It doesn't seem to even start trying to update the watch.

Self-Perfection, 2023-06-25:
I had connection issues after update 1.12 → 1.13 similar to [issues after 1.11 → 1.12 update](https://github.com/InfiniTimeOrg/InfiniTime/issues/1724#issuecomment-1511615600)

GadgetBridge just could not connect to the device. Well I know the drill now, so I deleted the device connection from Android settings again but could not pair it back from android bluetooth dialog even after several attempts. But adding the device back from GadgetBridge worked. I suggest everyone who have connection issues after 1.12 → 1.13 update try to do the following:
1. GadgetBridge: Hamburger menu → Data management → Export data
2. Remove pairing from Android settings (probably unnecessary)
3. Delete device from GadgetBridge
4. GadgetBridge: Hamburger menu → Connect new device → [perform pairing]
5. GadgetBridge: Hamburger menu → Data management → Import DB

Yet this is not a nice user experience. Prior to 1.12 OTA updates just worked without all this rigamarole.

hawkinspeter, 2023-06-25:
My issue is not the connecting per se, but trying to upload the 1.13 firmware. After it disconnects, I can just reconnect, but the watch is still on 1.12.

Bugsbane, 2023-06-25:
I rebooted the watch and then tried again and it actually updated to 1.13 fine. I wonder if it's some issue with needing to have rebooted since 1.12 was first installed and booted?

ricsim, 2023-06-26:
Thanx.

That worked!

 
 

Gesendet: Sonntag, 25. Juni 2023 um 18:04 Uhr

ajira86, 2023-06-26:
> I could update with Gadgetbridge, but Bluetooth does not work.

I had the same issue between 1.11 to 1.12 and had to unpair and pair again my device

Bugsbane, 2023-06-29:
> I had the same issue between 1.11 to 1.12 and had to unpair and pair again my device

Yes, I've had to do that a couple of times, too. Not a lot, but maybe twice ever.

MrMjau, 2023-06-30:
Do you have to unzip the file by any chance? ( 1.13 )

closed 2023-06-30 by MrMjau

hawkinspeter, 2023-07-01:
I've got mine to update by using Siglo on a linux laptop (over bluetooth) instead of GadgetBridge on Android.